UTRA-ART/SLAM-1920 #21: Deploy the Navigation Stack on Caffeine.

chandra-gummaluru, 2019-11-11:


AbiJaya1998, 2019-11-13:
Subtasks for this include 
- Potentially flash jetson with ubuntu 16.04 image 
- Potentially download ros kinetic on jetson board
- Make things work

Federico-PizarroBejarano, 2019-11-14:
@chandra-gummaluru I don't think we're ready for this ticket. It requires a lot of making sure we can get odometry from the robot to ros to our code, configuring things, checking the lidar, etc. I don't think the physical robot will be moving for a while. If anything, this ticket should be split into a ton of smaller ones, like making sure we get odom info, making sure we can give commands to the motors, making sure we get lidar information, etc. Just connecting the jetson and the motors and our code and viewing whats happening is going to take a long time, and I won't be physically there this Saturday so I can't help

AbiJaya1998, 2019-11-14:
I'll still work on it this Saturday ......I think it really depends on where electrical is at with their stuff ........... I don't think its a problem (in fact I think we should as long as Caffeine is reasonably ready ) to start this ticket, I can test some components individually if everything is not set up  Let me work on it a bit this Saturday If It becomes too much then I'll let u guys know and we can split this ticket up @chandra-gummaluru @Federico-PizarroBejarano

chandra-gummaluru, 2019-11-14:
Sounds like a plan. Trudie will also be joining us on Saturday (hopefully as a permanent member of SLAM), so maybe you could work with her on it since she'd have much more insight into the electrical side.

chandra-gummaluru, 2019-11-14:
> @chandra-gummaluru I don't think we're ready for this ticket. It requires a lot of making sure we can get odometry from the robot to ros to our code, configuring things, checking the lidar, etc. I [...]

This is a great point - my bad for overlooking it. Perhaps we could assign a small team to focus on the physical side of things, and create incremental tickets as we did with the simulated robot? Let's sync up and talk about this.

chandra-gummaluru, 2019-11-28:
Closed. Replaced with #32.

closed 2019-11-28 by chandra-gummaluru