UTRA-ART/SLAM-1920 #65: Ensure Caffeine + Husky works with all our Packages

erickmu1, 2020-01-19:
**Description**:
- Now that Navigation with LIDAR is coming to a close, we need to ensure all our packages are structured correctly and work both with Husky and Caffeine (in simulation and reality)
- **NOTE:** Given the open ended-ness of this task, add new Goals (issues to solve) below as they appear

**Goals**: (add as more issues come to light)
- [ ] Correct map update rate (observed in Real run)

**Acceptance Criteria**:
- [ ] Husky simulation runs Gmapping, Nav Stack, and `robot_localization` correctly with no outstanding errors/warnings
- [ ] Caffeine simulation runs Gmapping, Nav Stack, and `robot_localization` correctly with no outstanding errors/warnings
- [ ] Caffeine physical robot runs Gmapping, Nav Stack, and `robot_localization` correctly with no outstanding errors/warnings
- [ ] Packages are structured in a consistent fashion with a clear modular design

**Notes**:
- This issue is open ended as to what changes need to be made but should account for enabling packages to be swapped with little has and be easy to maintain
- Modular design ensures that adding a new "parameter" minimizes the number of files to be changed

AbiJaya1998, 2020-01-27:
Useful links - 

- https://answers.ros.org/question/232794/can-i-use-robot_localization-package-with-only-imu-data-if-i-couldhow-to-config-the-package/?fbclid=IwAR3GRRWsmc-XAUDfkJ0w4BpsmkeCT-1Bxf1fqcW0JxHACSsbonIBxIIhd60

- https://answers.ros.org/question/216856/robot_localization-drift/?fbclid=IwAR0sIqdQzK2exF4PeOfoiylLa_oQepx7UG4_yfxc0tx-EQ10aZjnwQZoHZk

AbiJaya1998, 2020-01-27:
Imp observations 
- Use both odom and IMU firstly 

[image]

AbiJaya1998, 2020-01-28:
More Useful links -
https://answers.ros.org/question/229722/how-to-stop-gazebo-publishing-tf/

closed 2020-02-22 by erickmu1